FooDeas/raspberrypi-ua-netinst #218: Reproducible Installs

johr, 2021-04-19:
I already used raspberrypi-ua-netinst succesfully. Really fine. 
What I wonder is, how could I get identical installs if I put the same installer-image on the SD-card, even if there are newer packages available? 
Sure, I can do an install, make an image of it and use that as base. But I want to use the installer, cause it is a much smaller image and being able to somehow fix the packge versions to use.
Any ideas on this?

Mausy5043, 2021-04-24:
For security reasons it is not advisable to pin packages on a given version, unless updating them to their current version would cause stability/usability problems. From a reproducibility point-of-view I do understand that you'd want to do that but it is not a wise choice. And from a stability point-of-view I see no reason why as I think the current Raspberry Pi OS is very stable indeed.  
I, personally, tend to re-build my Pies every 2 to 3 months from a freshly created installer and I keep all of them up-to-date by running `apt update; apt upgrade` once a week. 

Having said that, if you really want to pin package version you can start reading here: https://wiki.debian.org/AptConfiguration   

Beware, there be dragons!

johr, 2021-04-26:
Thanks. I understand. Sure updating on a regular basis makes sense and is planed. But the time when this happens, I was thinking about to have under my own control. 
But as you say, if raspbian is really stable then automatic updates, by the installer or through apt would not break anything.
Lets see,... maybe I give it a try:

closed 2021-04-26 by johr