FooDeas/raspberrypi-ua-netinst #219: Replace existing system

johr, 2021-04-19:
There is a section in the README https://github.com/FooDeas/raspberrypi-ua-netinst#reinstalling-or-replacing-an-existing-system
But  I did not find notes on how to replace an existing system. Is this even possible? Which steps to take?
Thanks.

kpfleming, 2021-04-19:
There are no special instructions for replacing an existing system; just start the installer as you normally would, and it will overwrite the installed system.

johr, 2021-04-21:
thanks. 
on the existing system I removed all files from the /boot folder and copied over the content of the installer-zip file (took care to not delete the initrd and kernel currently used by the system). 
reboot and system was installed :-)

closed 2021-04-21 by johr