FooDeas/raspberrypi-ua-netinst #221: How to enable Bluetooth

Mausy5043, 2021-07-04:
I'm trying to get Bluetooth to work.  
 
I followed the discussion in #20 but still struggled to enable BT on one of my RPi's.
Then discovered that #207 disables BT by default with commit [e958abc](https://github.com/FooDeas/raspberrypi-ua-netinst/pull/207/commits/e958abcfba8624c7b4bb290cf6bc6cb6d4ed3d96).

I can comment out the `dt-overlay=disable-bt` lines in `/boot/config.txt`.
Is `enable_uart=1` needed to make BT work or fix other issues that might arise?

FooDeas, 2021-07-04:
Removing both makes things easier on RPis model 3.

Mausy5043, 2021-07-04:
Is there a reason BT is disabled by default?

FooDeas, 2021-07-04:
The reason is documented in https://github.com/FooDeas/raspberrypi-ua-netinst/pull/207/commits/e958abcfba8624c7b4bb290cf6bc6cb6d4ed3d96. It should be no issue for most users.

Mausy5043, 2021-07-06:
So, most users prefer to use a serial console above BT. Didn't know that. Thanks.

My questions is answered and BT is working now. Closing this issue.

closed 2021-07-06 by Mausy5043