FooDeas/raspberrypi-ua-netinst #224: Raspbian bullseye - Installation fails "Iek!"

mpmc, 2021-09-15:
I decided to do a bit of testing and while an in place upgrade from buster to bullseye does work[1]. a fresh install does not and bails with `W: parser_rfc822: Iek! Don't find end of value!`

Full log from telnet...

```
Installation started at Wed Sep 15 15:27:27 UTC 2021 (UTC).

Resulting installer configuration:
  preset = server
  packages = nano
  firmware_packages = 
  mirror = http://raspbian.raspberrypi.org/raspbian/
  mirror_cache = 
  release = bullseye
  hostname = rpi-e1
  domainname = 
  user_ssh_pubkey = <REDACTED BY GCHQ>
  root_ssh_pubkey = 
  root_ssh_pwlogin = 1
  ssh_pwlogin = 
  username = mark
  usergpio = 
  usergpu = 1
  usergroups = 
  usersysgroups = 
  userperms_admin = 1
  userperms_sound = 0
  cdebootstrap_cmdline = --flavour=minimal --include=locales,console-setup,kmod,libpam-systemd,rng-tools,cpufrequtils,openssh-server,dosfstools,ntp,fake-hwclock,ifupdown,net-tools,vim-tiny,iputils-ping,wget,ca-certificates,rsyslog,cron,dialog,locales,tzdata,less,man-db,logrotate,bash-completion,console-setup,apt-utils,systemd-sysv,sudo,isc-dhcp-client
  packages_postinstall = nano,raspberrypi-bootloader,raspberrypi-kernel,raspberrypi-sys-mods,libraspberrypi-bin,raspi-copies-and-fills
  boot_volume_label = 
  root_volume_label = 
  bootsize = +128M
  bootoffset = 8192
  rootsize = 
  timeserver = time.nist.gov
  timezone = Europe/London
  rtc = 
  dt_overlays = 
  keyboard_layout = gb
  locales = en_GB.UTF-8
  system_default_locale = en_GB.UTF-8
  wlan_country = 
  ip_ipv6 = 1
  cmdline = dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 elevator=deadline fsck.repair=yes
  drivers_to_load = 
  gpu_mem = 32
  console_blank = 
  hdmi_type = 
  hdmi_tv_res = 1080p
  hdmi_monitor_res = 1024x768
  hdmi_disable_overscan = 0
  hdmi_system_only = 0
  usbroot = 0
  usbboot = 0
  rootdev = /dev/mmcblk0
  rootpartition = /dev/mmcblk0p2
  rootfstype = ext4
  rootfs_mkfs_options = 
  rootfs_install_mount_options = noatime,data=writeback,nobarrier,noinit_itable
  rootfs_mount_options = errors=remount-ro,noatime
  final_action = reboot
  quiet_boot = 0
  disable_raspberries = 0
  disable_splash = 0
  cleanup = 0
  cleanup_logfiles = 0
  spi_enable = 0
  i2c_enable = 0
  i2c_baudrate = 
  sound_enable = 0
  sound_usb_enable = 0
  sound_usb_first = 0
  camera_enable = 0
  camera_disable_led = 0
  use_systemd_services = 0

OTP dump:
  16:00280000
  17:1020000a
  18:1020000a
  27:00005050
  28:b9dd1201
  29:4622edfe
  30:00a01041
  45:472c0309

Waiting 5 seconds.....
Waiting for /dev/mmcblk0... 
  Disk /dev/mmcblk0: 59.7 GiB, 64076382208 bytes, 125149184 sectors
  Disklabel type: dos
  Disk identifier: 0x30d03f0b
OK
Applying new partition table... OK
Initializing /boot as vfat... OK
Copying /boot files in... OK
Initializing / as ext4... mke2fs 1.44.5 (15-Dec-2018)
ext2fs_check_if_mount: Can't check if filesystem is mounted due to missing mtab file while determining whether /dev/mmcblk0p2 is mounted.
  Discarding device blocks: done                            
  Creating filesystem with 15609856 4k blocks and 3907584 inodes
  Filesystem UUID: 12614d1e-2ab8-4818-8616-d2802d8e47a2
  Superblock backups stored on blocks: 
  	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
  	4096000, 7962624, 11239424
  
  Allocating group tables: done                            
  Writing inode tables: done                            
  Creating journal (65536 blocks): done
  Writing superblocks and filesystem accounting information: done   
  
OK
Mounting new filesystems... OK
Starting HWRNG... OK

Starting install process...
  P: Retrieving Release
  P: Retrieving Release.gpg
  P: Validating Release
  I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
  P: Parsing Release
  P: Retrieving Packages.xz
  P: Validating Packages.xz
  P: Parsing Packages
  W: parser_rfc822: Iek! Don't find end of value!
  E: Internal error: download

  ERROR: 1, trying again (2/5)...
  P: Retrieving Release
  P: Retrieving Release.gpg
  P: Validating Release
  I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
  P: Parsing Release
  P: Validating Packages
  P: Parsing Packages
  W: parser_rfc822: Iek! Don't find end of value!
  E: Internal error: download

  ERROR: 1, trying again (3/5)...
  P: Retrieving Release
  P: Retrieving Release.gpg
  P: Validating Release
  I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
  P: Parsing Release
  P: Validating Packages
  P: Parsing Packages
  W: parser_rfc822: Iek! Don't find end of value!
  E: Internal error: download

  ERROR: 1, trying again (4/5)...
  P: Retrieving Release
  P: Retrieving Release.gpg
  P: Validating Release
  I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
  P: Parsing Release
  P: Validating Packages
  P: Parsing Packages
  W: parser_rfc822: Iek! Don't find end of value!
  E: Internal error: download

  ERROR: 1, trying again (5/5)...
  P: Retrieving Release
  P: Retrieving Release.gpg
  P: Validating Release
  I: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
  P: Parsing Release
  P: Validating Packages
  P: Parsing Packages
  W: parser_rfc822: Iek! Don't find end of value!
  E: Internal error: download

  ERROR: 1

Error: The installation could not be completed!
  2 retries left.
  You have 10 seconds to press ENTER to get a shell or it will be retried automatically.

```

Would be awesome if we could get this fixed. Thanks

[1]: On an existing fresh install of buster - I just edited apt sources to point to bullseye from the raspberrypi.org mirror, and did the usual update/upgrade/dist-upgrade etc dance. Rebooted OK & came up no issues.

Bebop182, 2022-01-21:
I'm interested in this aswell

Mausy5043, 2022-01-21:
I'm stuck with this. The workaround that works for me is using the installer with `buster` and then upgrade using #225 (don't forget to look at my last comment though!).

PUsmanKhan, 2022-12-15:
When are we having an official release for bullseye?

ambientsound, 2023-06-19:
This is such a great project that has been useful to me for so many years, I really hope to see bullseye support soon. Cheers :+1:

FooDeas, 2023-06-25:
Will be fixed in the upcoming release!

closed 2023-06-25 by FooDeas