FooDeas/raspberrypi-ua-netinst #225: Please add support for bullseye

miroslavbucek, 2021-10-30:
[image]

Mausy5043, 2021-11-01:
Starting from an installed system on `buster` you can use this workaround awaiting support by the installer:

```(bash)
#!/bin/bash

# Set this to the current release you are on
thisrelease=buster
# Set this to the release you wish to migrate to
nextrelease=bullseye

# make sure we are current.
sudo apt-get update
sudo apt-get -y dist-upgrade

# source: https://linuxconfig.org/raspbian-gnu-linux-upgrade-from-jessie-to-raspbian-stretch-9
dpkgC=$(sudo dpkg -C | wc -l)
if [ $dpkgC -ne 0 ]; then
  sudo dpkg -C
  echo
  echo "*** Consistency check returned a list of partially installed, missing and/or obsolete packages."
  echo "*** Repair these issues before continuing."
  exit 1
fi

aptholds=$(sudo apt-mark showhold | wc -l)
if [ $aptholds -ne 0 ]; then
  sudo apt-mark showhold
  echo
  echo "*** There are packages ON HOLD."
  echo "*** Repair these issues before continuing."
  exit 1
fi

echo
# Point to the new repository
sudo sed -i "s/${thisrelease}/${nextrelease}/g" /etc/apt/sources.list
cat /etc/apt/sources.list
echo
sudo sed -i "s/${thisrelease}/${nextrelease}/g" /etc/apt/sources.list.d/raspberrypi.org.list
cat /etc/apt/sources.list.d/raspberrypi.org.list
echo

echo
echo "***"
echo "*** UPGRADING FROM ${thisrelease} TO ${nextrelease}"
echo "*** THIS WILL TAKE A FEW MINUTES"
echo "***"

# allow user to change their mind...
sleep 10

sudo apt-get update
sudo apt-get -y dist-upgrade
```

miroslavbucek, 2021-11-02:
I tried the manual upgrade from buster but it failed. The system booted but failed to log in, even remotely using SSH, failed to log in locally.

As you can see in the video.

Uploading IMG_0897.MOV…

Mausy5043, 2021-11-05:
Remarkable.

Mausy5043, 2021-11-05:
Hmmm. I may have mistyped.
Could you try adding `sudo apt -yuV full-upgrade` to the script so that the last three lines read:
```
sudo apt-get update
sudo apt -yuV full-upgrade
sudo apt-get -y dist-upgrade
```

See if that works.

miroslavbucek, 2021-11-19:
It works fine! Thanks

closed 2021-12-25 by miroslavbucek