FooDeas/raspberrypi-ua-netinst #226: Raspberry pi Zero 2 W support

ignaciojimenez, 2022-01-20:
Trying to make this work on a Zero 2 W I just bought it appears that nothing even happens. No logs, no activity. I will keep exploring but nothing appears to happen.

avian2, 2022-07-05:
The current v2.4.0+ release build is pretty old (Apr 2020) and hence contains firmware and kernel from an outdated Raspbian release. It does not boot on hardware that needs more recent firmware releases.

I've seen this on Compute Module 4 rev. 5 (released in Apr 2022) and I'm guessing it is also the reason Zero 2 W (released in Oct 2021) might not work with the current release.

Since the kernel does not even boot, there's nothing written to ua-netinst's log file. If you connect an HDMI display you might see the error message about the outdated firmware files (on CM4 rev. 5 it's only visible for a short while before it scrolls away) before the boot hangs.

My solution was to just rebuild ua-netinst from git using the instructions in the BUILD.md file. The build process pulls in the latest kernel and firmware and the re-built ua-netinst seems to work just fine with new hardware.

ignaciojimenez, 2022-07-05:
It was clear to me that the project is not actively maintained, so I decided to migrate my provisioning and orchestration to pxe+ansible. It is still a nice project and I will keep an eye on it. Thanks for checking yourself and answering as well :)

FooDeas, 2023-08-15:
Please check v2.5.0.

closed 2023-08-15 by FooDeas