FooDeas/raspberrypi-ua-netinst #229: Bookworm support

chris18890, 2023-10-13:
A new version of Raspberry Pi OS based on Debian 12 (bookworm) as been released!

See here for the release announcement - https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

Given the number of changes between Bullseye & Bookworm, an upgraded is actively discouraged & the recommended way is to flash a new Bookworm image ontot he SD Card

What work needs to be done to enable raspberrypi-ua-netinst to be able to install Bookworm?

Phiber2000, 2023-10-17:
First steps identified when I had a first look on it a few weeks ago:
- add Debian archive keys for v12
- can we remove the v11 keys? (identify issues)
- identify and get the "new" firmware packaged working in the boot partition

FooDeas, 2023-10-17:
First steps identified when I had a first look on it a few weeks ago:
- add Debian archive keys for v12
- can we remove the v11 keys? (identify issues)
- identify and get the "new" firmware packaged working in the boot partition

FooDeas, 2023-10-17:
I just pushed my [bookworm](https://github.com/FooDeas/raspberrypi-ua-netinst/tree/bookworm) branch as base and playground to fork and work on.

chris18890, 2023-10-17:
> First steps identified when I had a first look on it a few weeks ago:
> 
>     * add Debian archive keys for v12
> 
>     * can we remove the v11 keys? (identify issues)
> 
>     * identify and get the "new" firmware package working in the boot partition

That's the v10 key you've removed in that commit

FooDeas, 2023-10-18:
> That's the v10 key you've removed in that commit
I already removed the unneeded v10 key without negative impact. That's not related to the removal of the v11 key that I also wanted to remove and got problems with the v12 key verification (keychain issue? not proven yet.).

conversy, 2023-12-08:
Hello, I was wondering if there were any news on this, and maybe on rpi5 support? Best!

oupala, 2025-05-10:
I'd love to see this branch released as a GitHub release.

Would you mind publishing it?