FooDeas/raspberrypi-ua-netinst #230: Configuration to disable root

busmei, 2024-02-02:
Hi,

first of all many thanks for working on this project. I really appreciate your effort and dedication.

I was playing around with the config parameters in installer-config.txt. If understood the documentation correctly, it shall be possible to disable the root account by settings rootpw as well as root_ssh_pubkey empty. So this is what I set:

username=pi
userpw=test
userperms_admin=1
rootpw=
root_ssh_pubkey=
root_ssh_pwlogin=0

However, after completing the installation I was still able to login with root/raspbian. Please correct me if my understanding is wrong. Otherwise there may be a bug somewhere.

In the meantime I use a workaround with post-install.txt containing
sed -E "s/^([root:]+:)([^:]+)(..+)$/\1!\3/g" -i /rootfs/etc/shadow

Looking forward to your feedback.

Thanks again.

BR, Mike

TwinUsers, 2024-02-02:
Hi,

Take a look at scripts/opt/raspberrypi-ua-netinst/install.sh on line 155

variable_set "rootpw" "raspbian"

maybe this needs to be blank, so therefore you can't login

FooDeas, 2024-02-02:
@busmei Do you mean local or SSH login?

busmei, 2024-02-02:
@FooDeas local